mobizt/ESP_SSLClient #1: EthernetClient not connecting

kapyaar, 2023-07-29:
Hi,

I am testing https.ino example with esp32+w5500. While the code itself works if I use wifi only, when I switch to EthernetClient, after ethernet is initialized successfully, when the connection is attempted, it goes right away to 

---------------------------------
Connecting to server...ERROR - Generic error
 failed

I am using Ethernet2 library, and I have made some modifications such as max sockets=2, use large buffers, etc (as recommended for similar use cases). Have you seen such issues? I do see in the code that you have commented out EthernetClient base_client, so I am guessing you may have tested it? Much appreciate any suggestions. 

PS, the ESP32+w5500 works well for normal code. So, I know the hardware is good.

mobizt, 2023-07-31:
I test with my W5500 module and ESP32 and it works fine.

There are some user feedbacks about library when using ESP32 with W5500 both works and not work.

Especially when I implement it on my other library that supports external `SSLClient` and concluded that it may not work in all configurations even using `WiFIClient` as a client.

The problem is the SSL engine library used in this library for ESP32 is `mbedTLS` that is already come with core SDK installation.

The `mbedTLS` works well when binding with lwIP TCP socket directly as in WiFiClientSecure. 

The problem is not only at Ethernet module buffer memory restriction but the binding between `mbedTLS` and Client has some stability and slow data transfer speed issues.

When I test `mbedTLS` binding with Client (WiFiClient) without socket and Client (WiFiClient) with socket, the result is worst when compare with WiFiClientSecure.

Othe SSLClient libraries that use `mbedTLS` in the same approach as in this library will not work good in all conditions.

Now I decide to remove `mbedTLS` as a SSL library and use `BearSSL` SSL engine library instead as used in ESP8266 core.

From testing ESP32 with `BearSSL`, it works well as fast connection as in WiFIClientSecure that used `mbedTLS`.

I and other users also test ESP32 with `BearSSL` and W5500 and it works fine.

I will finish updating the library soon in this week.

I will post here when the new version is available.

kapyaar, 2023-07-31:
Thank you for the quick response. I tested with just wifi, and that works fine, I know that is using bearSSL. It would be great to try out your new version with bearSSL. Looking forward to it. :)

mobizt, 2023-08-01:
Now the new version ([v2.0.0](https://github.com/mobizt/ESP_SSLClient/releases/tag/v2.0.0)) is available.

Arduino's Library Manager may be available soon or download zip file to install.

kapyaar, 2023-08-01:
Did a quick test, and it works. Thanks, I will test further, and update you if there are any issues. Much appreciated :)

closed 2023-08-06 by mobizt