mobizt/ESP_SSLClient #3: JWT error

back920096, 2023-09-11:
Hello author, this problem occurs when reconnecting after a few days

Token error: code: -1, message: Invalid JWT: Token must be a short-lived token (60 minutes) and in a reasonable timeframe. Check your iat and exp values ​​in the JWT claim.

Will this problem occur if the gmtOffset setting of Firebase.setUDPClient does not match the current zone?
Or is there any part that needs special attention?

mobizt, 2023-09-11:
You should post in the Firebase repo.

First, you should update the library to latest version.

The Firebase function `setUDPClient` is now deprecated.

Normally you can set the time zone via `FirebaseConfig` i.e., `config.time_zone` and  `config.daylight_offset`

You can set your device time yourself before calling `Firebase.begin` too which in this case, library will not touch your device time.

closed 2023-09-11 by mobizt

back920096, 2023-09-11:
Thanks for the suggestion
After upgrading the version, I found some problems using the latest FCM
example.
fbdo.setEthernetClient(&eth, ETH_MAC, WIZNET_CS_PIN, WIZNET_RESET_PIN,
nullptr);
[image: image.png]
Problems using DHCP

Suwatchai K.

mobizt, 2023-09-11:
I did not see the image. Please post your error here.

mobizt, 2023-09-11:
Please go to GitHub and don't reply email directly.

back920096, 2023-09-11:
Thanks for the suggestion
After upgrading the version, I found some problems using the latest FCM example.
fbdo.setEthernetClient(&eth, ETH_MAC, WIZNET_CS_PIN, WIZNET_RESET_PIN, nullptr);
Problems using DHCP
[image]
[image]

mobizt, 2023-09-11:
See the example, ETH_MAC is actually `uint8_t` array, you can't use something like `std::initializer_list<uint8> `.

`uint8_t Eth_MAC[] = {0x02, 0xF0, 0x0D, 0xBE, 0xEF, 0x01};`

back920096, 2023-09-11:
Thanks to the author for the reminder

[image]
Token error: code: -115, message: NTP server time reading cannot begin when valid time is required because of no WiFi capability/activity detected. Please set the library reference time manually using Firebase.setSystemTime

Use esp32s3+w5500
I would like to ask if the new version uses external Client. Do I need to set anything in firebaseFS.h?
[image]

I only saw w5500 with esp8266
Didn't see support for esp32

mobizt, 2023-09-11:
The examples are [here](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/ExternalClient/RTDB/Ethernet).

> NTP server time reading cannot begin when valid time is required because of no WiFi capability/activity detected. Please set the library reference time manually using Firebase.setSystemTime

As the error shown, you use external client (Ethernet) to connect, and your WiFi is also off, library will not or unable to use your provided client to access NTP server to get the time as it required another UDP client which is now not included in this library.

It is your job to set your device time manually before calling `Firebase.connect`.

In some devices, user needs to set the library reference time to use internally via function `Firebase.setSystemTime` except for ESP8266 and ESP32 devices that library can take the time from its timer directly.,

mobizt, 2023-09-11:
The time `169445xxxx` is the epoch time of tomorrow.

mobizt, 2023-09-11:
```cpp
Firebase.setSystemTime(1694425572); // Current Unix epoch Time

Serial.println(Firebase.getCurrentTime());
Serial.println(time(nullptr));
```

mobizt, 2023-09-11:
The function `Firebase.setSystemTime` should return true for valid time set.

You should write your code carefully to prevent such human error.

back920096, 2023-09-11:
[image]
I use Firebase.setSystemTime which always returns 0
But Serial.println(Firebase.getCurrentTime());
     Serial.println(time(nullptr));
It shows that there is writing, but it still jumps. Token error: code: -115
[image]

mobizt, 2023-09-11:
Ok I will check this and update.

mobizt, 2023-09-11:
Now you can update library to v4.4.2 with this issue fixed.

back920096, 2023-09-12:
Thanks to the author. Successfully connected.
[image]
[image]
Is there any way to avoid clogging after unplugging the network cable?
I have tried using Firebase.reconnectWiFi(false) and it still gets blocked.

back920096, 2023-09-12:
[image]
[image]
[image]
Report an error using fbdo.setExternalClient

mobizt, 2023-09-12:
It quite off topic here.

When using `setEthernetClient` There is 60 seconds waiting time for DHCP as long as `Firebase.ready()` was executed. 

The function `reconnectWiFi` is applied only for WiFi and external client with network connection callback function.

The network connection callback function should do appropriate network disconnection first.

For GSM and Ethernet integration (`setEthernetClient`  and `setGSMClient`), the network reconnection was self-controlled and done automatically without user consent which is the library purpose. 

You have two choices in your case.

1. Use `setGenericClient` with callbacks (formerly `setExternalClient` and callbacks)
2. Temporary exclude `Firebase.ready()` from loop and pause the FirebaseData object with `pauseFirebase(true)` when you want to stop library to connect to server.

### For Client
The client is generic/basic/plain network Arduino Client instead of SSL Client.

Please see the Readme and examples for new version.

mobizt, 2023-09-12:
> Report an error using fbdo.setExternalClient

Thanks for the bug report.  I will check and update.

back920096, 2023-09-12:
fbdo.setGenericClient has the same error
[image]
[image]
[image]

mobizt, 2023-09-12:
I know it.

mobizt, 2023-09-12:
Now you can update the library to fix the issue.

back920096, 2023-09-12:
Already running normally
I'm happy to help you find the bug

mobizt, 2023-09-12:
Thanks, the issue is usually found as the main library core has changed with huge improvement in memory and performance in v4.4.x.