mobizt/ESP_SSLClient #4: Data loss

DriekdeGadgetfreak, 2024-01-16:
As a test I replaced:

WiFiClientSecure client;
client.setInsecure();

by

ESP_SSLClient client;
WifiClient net;
client.setInsecure();
client.setClient(&net);

I connect to a website which returns 7712 (= content-length) bytes with the original code. 
The replaced code gives a read timeout after 6703 bytes, with the same content-length in the return header.
Do I do something wrong?

Regards, Hans

mobizt, 2024-01-16:
If you read the Readme, you have to set the BearSSL IO reserve memory size that large enough for the SSL data especially Rx buffer size.

>   // Set the receive and transmit buffers size in bytes for memory allocation (512 to 16384).
>   // For server that does not support SSL fragment size negotiation, leave this setting the default value
>   // by not set any buffer size or set the rx buffer size to maximum SSL record size (16384) and 512 for tx buffer size.  
>   ssl_client.setBufferSizes(1024 /* rx */, 512 /* tx */);

You can set it up to 16k for full SSL data frame.

closed 2024-01-16 by mobizt

DriekdeGadgetfreak, 2024-01-16:
Thanx! That helps me.