mobizt/ESP_SSLClient #5: SSL connect hangs

SandroSartoni, 2024-02-17:
Hi,

I am using the ESP_SSLClient library on a LilyGo TCALL v1.4 with a SIM800H module mounted inside.
The project I am working on measures a few parameters (air humidity, temperature and soil moisture), then sends them to a server through HTTPS. I can only rely on GSM, since the place where this board will be installed does not have access to WiFi.

Below is the code I have written:
```
#include "DHT.h"
#include <esp_task_wdt.h>

//////////////////////////////
// Sensor Variables Section //
//////////////////////////////

#define WDT_TIMEOUT   6*60  // Watchdog for three minutes
#define DHTTYPE       DHT11 // We're using DHT 11 for this project
#define DHTPIN        15    // Serial communication for DHT sensor goes through this pin
#define SOILSENS      34    // The analog signal from the soil mosture sensor is fed in this pin
#define MEASUREMENTS  5     // How many measurements should be computed in a measurement cycle

// Define the dht variable holding type and pin information
DHT dht(DHTPIN, DHTTYPE);

// Temperature and air humidity variables are defined here
float temperature[MEASUREMENTS];
float air_humidity[MEASUREMENTS];

// Soil moisture variables defined here
const int air_moisture = 3390;            // This is the minimum value we can get through the soil moisture sensor
const int water_moisture = 1490;          // This is the maximum value we can get through the soil moisture sensor
int soil_moisture_val;                    // This variable will then be scaled wrt the air and water moisture levels
int soil_moisture_percent[MEASUREMENTS];  // Final moisture value holding array

////////////////////////////////
// Internet Variables Section //
////////////////////////////////
// GPRS credentials (leave empty, if not needed)
const char apn[]      = "TM"; // APN (example: internet.vodafone.pt) use https://wiki.apnchanger.org
const char gprsUser[] = "";   // GPRS User
const char gprsPass[] = "";   // GPRS Password

// SIM card PIN (leave empty, if not defined)
const char simPIN[]   = "1503"; 

// Server details
// The server variable can be just a domain name or it can have a subdomain. It depends on the service you are using
const char server[] = <my_server>;   // domain name: example.com, maker.ifttt.com, etc
const int  port = 443;                                                   // server port number

const char token [] = <my_token>;

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22

// Set serial for debug console (to Serial Monitor, default speed 115200)
#define SerialMon Serial
// Set serial for AT commands (to SIM800 module)
#define SerialAT Serial1

// Configure TinyGSM library
#define TINY_GSM_MODEM_SIM800      // Modem is SIM800
#define TINY_GSM_RX_BUFFER   1024  // Set RX buffer to 1Kb

#include <Wire.h>
[... 355 lines cut ...]
  delay(5000);

  Serial.println("[LOOP] Device now is going to sleep...");
  
  // This will put the ESP32 to sleep
  esp_deep_sleep_start();

}
```

The issue I am encountering is that from time to time the *connectSSL()* function fails, and that can happen multiple times in a row thus stalling the whole transmission process.
Setting the debug option to 4, this is what I read when such failures occur:
> INFO.mConnectBasicClient: Basic client connected!
> INFO.mConnectSSL: Start connection.
> INFO.mConnectSSL: Wait for SSL handshake.
> INFO.mUpdateEngine: State RECVREC
> INFO.mUpdateEngine: State RECVREC
> INFO.mRunUntil: SSL state changed.
> INFO.mRunUntil: State RECVREC
> INFO.mRunUntil: Expected bytes count: 5
> ERROR.mRunUntil: SSL internals timed out!
> ERROR.mConnectSSL: Failed to initlalize the SSL layer.
> ERROR.mConnectSSL: Unknown error code.

Is it something related to my code/settings? How can I fix this problem? Sometimes the board will be able to connect at the first try and everything goes smoothly, sometimes it takes few retries and other times is completely stuck. I am using the latest version of this library.

mobizt, 2024-02-17:
You may have to increase TINY_GSM_RX_BUFFER in GSM library.

The rx buffer setting via `setBufferSizes` should be larger than your data + some cipher suite.

This depends on the fragmentation supports on server.

If server does not support fragmentation, this buffer should set to 16k.

closed 2024-02-17 by mobizt

SandroSartoni, 2024-02-18:
Hi @mobizt, I have updated both TINY_GSM_RX_BUFFER and setBuffersizes function to have 16384 bytes for the RX buffer but the error still occurs. It does not happen for every transmission, sometimes it goes through just fine while other times it shows the error I reported in the previous post.

mobizt, 2024-02-18:
You're starting SSL connection (SSL handshake) without stopping the current connection which there are some data that is waiting for read.

You have to review your code logic to make it work properly.

SandroSartoni, 2024-02-18:
Thank you for your reply. I am not sure what you mean by stopping the current connection: the code I reported is taken from one of the sketch examples reported in the library, the "http upgrade" one.

mobizt, 2024-02-18:
I mean session or TCP session hope you understand.

The session was not re-used in the examples, and it works normally.

The error showed that your start a new connection while the session was opened and the BearSSL engine is in writing or reading for incoming data state. You have to terminate the sever connection before starting new connection.

This is your code logic and timing issue.

mobizt, 2024-02-18:
The http upgrade is not needed for normal application unless IMAP and SMTP applications.

For GSM module, the simple example is here.
https://github.com/mobizt/ESP_SSLClient/blob/main/examples/gsm/gsm.ino