mobizt/ESP_SSLClient #7: Allow custom Port

rjjrbatarao, 2024-06-20:
Hi thanks for the example earlier, I finally saw the culprint to my problem its mIsSecurePort allows only predefined secure ports in my case Im using a custom port like 5353 or 5443, I added my custom port on the ESP_SSLClient_Const.h _secure_ports and finally it worked!!. Maybe you can provide a flag or method to turn off the checking?, THank you.

mobizt, 2024-06-20:
Your request ports are not standard secure port. 

It needs more works to support user defined port more than allowing connection but the types of ports (Plain, SSL, TLS, STARTTLS) should be defined by user together with the port number.

The email client library that uses this ssl client allows user to connect to custom IMAP and SMTP ports.

This is not hard to implement but it needs the time, and I will update here when it becomes available.

mobizt, 2024-06-20:
I think that I don't have to allow non-standard secure ports as you can connect in plain text mode then upgrade as in [this example](https://github.com/mobizt/ESP_SSLClient/blob/main/examples/http_upgrade/http_upgrade.ino).

mobizt, 2024-06-20:
You don't have to modify the source when your port does not in the list, you can upgrade as example above.

This should be done in user code or application that use this library to manually upgrade to secure connection in case non-standard secure port as implemented in my email client app that does not touch the ssl client library source.

mobizt, 2024-06-20:
I close this issue now as the alternate option is available.

closed 2024-06-20 by mobizt

mobizt, 2024-06-20:
I test with any non-standard secure port e.g. 5443 by upgrading with `ssl_client.connectSSL()` and it works as expected.

rjjrbatarao, 2024-06-20:
Amazing!! Im now using it with my custom port without touching the source code, thank you for this awesome trick :)