mobizt/ESP_SSLClient #9: MQTT-TLS with Ethernet w5500

MaxiLargo, 2024-07-03:
Hello, I have a somewhat similar problem. I have to connect to a broker (this one gave me a certificate) and when I want to connect with ethernet w5500

```
EthernetClient ethClient;
ESP_SSLClient ssl_client; 
PubSubClient client(ssl_client);

void setup() {
  Serial.begin(115200);

M5.begin();
 M5.Power.begin();
 SPI.begin(SCK, MISO, MOSI, -1);
 Ethernet.init(CS);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // start the Ethernet connection:
  Serial.println("Initialize Ethernet with DHCP:");
  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
    // Check for Ethernet hardware present
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
      while (true) {
        delay(1); // do nothing, no point running without Ethernet hardware
      }
    }
    if (Ethernet.linkStatus() == LinkOFF) {
      Serial.println("Ethernet cable is not connected.");
    }
    // try to congifure using IP address instead of DHCP:
    Ethernet.begin(mac, ip);
  } else {
    Serial.print("  DHCP assigned IP ");
    Serial.println(Ethernet.localIP());
  }
   ssl_client.setCACert(root_ca);
    ssl_client.setBufferSizes(1024 /* rx */, 512 /* tx */);
    ssl_client.setDebugLevel(1);
    ssl_client.setClient(&ethClient);
  client.setServer(mqtt_server, 8883); // Puerto 8883 para conexión segura

}
```

and the logs was:

>Failed to connect to MQTT broker, rc=-2
>ERROR.mConnectSSL: Failed to initlalize the SSL layer.
>ERROR.mConnectSSL: Certificate is expired or not yet valid.

can you help me pls?

mobizt, 2024-07-03:
This library provides the same functions as ESP8266 `WiFiClientSecure`.

As @Pablo2048 said, you have to set the time for the X509 certificate verification which can do with function `ESP_SSLClient::setX509Time` or set your system time from NTP server via `configTime` or set manually via `settimeofday`.

MaxiLargo, 2024-07-03:
Ok, i put the time with an NTP Server with EthernetUDP, and error change

```
EthernetClient basic_client;
ESP_SSLClient ssl_client;

PubSubClient client(ssl_client);

const char* root_ca PROGMEM = R"EOF(...)

void setupTime() {
  timeClient.begin();
  timeClient.update();
  time_t now = timeClient.getEpochTime();
  Serial.print(now);
  ssl_client.setX509Time(now);
}
void setup() {
  Serial.begin(115200);

M5.begin();
  M5.Power.begin();
  SPI.begin(SCK, MISO, MOSI, -1);
  Ethernet.init(CS);
  while (!Serial) {
    ;
  }

  Serial.println("Initialize Ethernet with DHCP:");
  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
      while (true) {
        delay(1); 
      }
    }
    if (Ethernet.linkStatus() == LinkOFF) {
      Serial.println("Ethernet cable is not connected.");
    }
    Ethernet.begin(mac, ip);
  } else {
    Serial.print("  DHCP assigned IP ");
    Serial.println(Ethernet.localIP());
  }
      M5.Lcd.println("M5Stack W5500 Test");
    M5.Lcd.println(" ");
    M5.Lcd.print(Ethernet.localIP());
 
 ssl_client.setCACert(root_ca);

  ssl_client.setBufferSizes(1024, 1024);

    ssl_client.setDebugLevel(1);

    ssl_client.setClient(&basic_client);
    setupTime();
    
  client.setServer(mqtt_server, 8883); // Puerto 8883 para conexión segura

}
```
logs:
> Failed to connect to MQTT broker, rc=-2
> ERROR.mConnectSSL: Failed to initlalize the SSL layer.
> ERROR.mConnectSSL: Chain could not be linked to a trust anchor.

mobizt, 2024-07-03:
The `root_ca` you set is not the root certificate of mqtt server.

MaxiLargo, 2024-07-03:
I used the same `root_ca` for other projects with `WiFiClientSecure` and had no problems publishing and subscribing from broker, the root certificate is from HiveMQ

closed 2024-07-03 by mobizt