mobizt/ESP_SSLClient #10: MQTT-TLS with Ethernet w5500

MaxiLargo, 2024-07-12:
Hi it's me again. Finally i can connect me to the broker mqtt successfully, but in another proyect [https://github.com/mobizt/ESP_SSLClient/issues/9#issue-2388394631](url) .

In my main proyect, i try to use the same but i had some problems:

```
void setup() {
  WiFi.mode(WIFI_STA);  // explicitly set mode, esp defaults to STA+AP

  Serial.begin(115200);

  disableCore0WDT();
  disableCore1WDT();
  M5.begin();        
  M5.Power.begin();
    
  uint64_t chipId1 = 0;
  chipId1 = ESP.getEfuseMac();

  char chipIdString[18];
  snprintf(chipIdString, sizeof(chipIdString), "%04X%08X", (uint16_t)(chipId1 >> 32), (uint32_t)chipId1);
  Serial.print(chipIdString);
  
  if (!SD.begin()) {
        M5.Lcd.println("Card failed, or not present");
        while (1);
    }
    M5.Lcd.println("TF card initialized.");

    while (!module.begin(&Wire, 21, 22, MODULE_4IN8OUT_ADDR)) {
        Serial.println("4IN8OUT INIT ERROR");
        M5.Lcd.println("4IN8OUT INIT ERROR");
    }
    
  SPI.begin(SCK, MISO, MOSI, -1);
  Ethernet.init(CS);

  Ethernet.begin(mac);
  Serial.print("Ethernet iniciado. Dirección IP: ");
  Serial.println(Ethernet.localIP());
  timeClient.begin();
  timeClient.update(); 

   time_t currentTime = timeClient.getEpochTime();

    ssl_client.setX509Time(currentTime);

    ssl_client.setCACert(rootCA);

    ssl_client.setBufferSizes(1024 , 1024 );

    ssl_client.setDebugLevel(1);

    ssl_client.setClient(&basic_client);
  Reinicio.attach(2,reinicio);

  xTaskCreatePinnedToCore(
    mqttTask,   
    "mqttTask",  
    8192,         
    NULL,
    1,  
    NULL,
    0   
  );
  xTaskCreatePinnedToCore(
    countTask,    
    "countTask",  
    8192,         
    NULL,
    1,  
    NULL,
    1   
  );

  client.setServer(mqtt_server, mqtt_port);  
  client.setCallback(callback);

 
  M5.Lcd.fillScreen(ILI9341_BLACK);
  M5.Lcd.setTextColor(ILI9341_YELLOW, ILI9341_BLACK);  
  M5.Lcd.fillRect(0, 0, 320, 30, ILI9341_NAVY);
  M5.Lcd.setTextColor(ILI9341_WHITE);
  M5.Lcd.setTextSize(2);
  M5.Lcd.setCursor(15, 6);  
  M5.Lcd.print("Registro produccion V2.5");
  M5.Lcd.setCursor(10, 40); 
  M5.Lcd.print("Maquina:");
  M5.Lcd.setCursor(10, 70); 
  M5.Lcd.print("Contador:");
  M5.Lcd.setCursor(10, 180); 
  M5.Lcd.print("Operario:");
  M5.Lcd.setTextSize(2);
  M5.Lcd.setCursor(10, 210); 
  M5.Lcd.print("IP:");
  M5.Lcd.setCursor(10, 150);
  M5.Lcd.print("Conexion:");
  M5.Lcd.setCursor(180, 70);
  M5.Lcd.print("Scrap:");
  M5.Lcd.setCursor(45, 210);
  M5.Lcd.setTextColor(ILI9341_YELLOW, ILI9341_BLACK);
  M5.Lcd.print(Ethernet.localIP());
  valor.store(readFile(SD, "/inputInt.txt").toInt());

}
```
and the log was:

 > ERROR.write: Failed while waiting for the engine to enter BR_SSL_SENDAPP.
 > ERROR.available: Cannot operate on a closed SSL connection.

Pd: In the main proyect i use 2 task, 1 for mqtt and another for an oled screen

mobizt, 2024-07-12:
You have to use `setSessionTimeout` in case the TCP session was kept alive while server terminates the connection.
https://github.com/mobizt/ESP_SSLClient?tab=readme-ov-file#set-the-tcp-session-timeout-in-seconds

You have to try some timeout e.g. 60 to 180 seconds depends on its server policy.
e.g.

`ssl_client.ssl_client1.setSessionTimeout(150); // 150 sec or 2.5 min timeout`

closed 2024-07-12 by mobizt

mobizt, 2024-07-12:
The `setSessionTimeout` is the new feature and you should update the library to [v2.1.9](https://github.com/mobizt/ESP_SSLClient/releases/tag/v2.1.9).

The timeout will be checked since last connect and last write.