mobizt/ESP_SSLClient #12: GSM.ino code not connecting to server this example have issues

Manojkumar0308, 2024-07-24:
11:49:24.744 -> [4258] Wait...
11:49:27.734 -> [7258] Initializing modem...
11:49:27.734 -> [7259] ### TinyGSM Version: 0.12.0
11:49:27.734 -> [7259] ### TinyGSM Compiled Module:  TinyGsmClientSIM7600
11:49:29.914 -> [9438] ### Modem:  
11:49:29.914 -> [9438] ### Modem:  
11:49:41.465 -> [21005] ### Unhandled: SMS DONE
11:49:41.465 -> 
11:49:41.465 -> +CGEV: EPS PDN ACT 1
11:49:41.465 -> [21005]  setNetworkMode faill
11:49:41.465 -> [21015] ### Modem: INCORPORATED A7672S-LASC
11:49:41.491 -> [21015] Modem Name: INCORPORATED A7672S-LASC
11:49:41.491 -> [21044] Modem Info: Manufacturer: INCORPORATED Model: A7672S-LASC Revision: A7672M6_V1.11.1 IMEI: 864180054656835 +GCAP: +CGSM,+FCLASS,+DS
11:49:41.533 -> Waiting for network... success
11:49:41.533 -> Network connected
11:49:41.533 -> Connecting to  success
11:49:41.565 -> GPRS connected
11:49:41.565 -> ---------------------------------
11:49:41.601 -> Connecting to server...> ERROR.mConnectBasicClient: Failed to connect to server using basic client.
11:49:41.643 ->  failed
11:49:41.643 -> 
11:49:41.643 -> 
11:49:41.643 -> GPRS disconnected

mobizt, 2024-07-24:
It is your code fault as you can see the error that the GSM client cannot connect to the network.

ERROR.mConnectBasicClient: Failed to connect to server using basic client.

You should check your code and follow the library examples.

closed 2024-07-24 by mobizt

Manojkumar0308, 2024-07-24:
I didn't did any thing with your example I just copied it and paste it and
i got

failed to connect to server

mobizt, 2024-07-24:
Check the server you are connecting to.

It should exist.

The error is from TinyGSMClient library not a  SSL client library error.

You have to test your server with TinyGSM library alone before using this library.