mobizt/ESP_SSLClient #16: Inconsistant Behaviour for  using GSM (Sim800l)

SherifMEldeeb, 2024-08-24:
I am having problem connecting to mqtts/https using tls over GSM
My setup is:
micro-controller: ESP32
communication: TinyGsmClient for Sim800l
TLS Client:  mobizt/ESP_SSLClient
Cert: Let's encrypt
Note that same codebase everything works by changing 
communication: WiFiClient 

the problem is the connection is not stable sometimes it disconnects while handshake with different errors like:
1-
ERROR.mConnectSSL: Failed to initlalize the SSL layer.
ERROR.mConnectSSL: SSL received fatal alert - Key type does not match that which was expected.
2-
ERROR.mConnectSSL: Failed to initlalize the SSL layer.
ERROR.mConnectSSL: SSL received fatal alert - Decoding error: extraneous element.
3- (this one after connected)
ERROR.available: Cannot operate on a closed SSL connection.
ERROR.available: SSL received fatal alert - Server sent an extension type that we did not announce, or used the same extension type several times in a single ServerHello.

when disconnects at handshake i get several notices on mqtt logs like:
1-
ssl_handshake:verify_connection/6:484: TLS server: In state cipher at ssl_handshake.erl:484 generated SERVER ALERT: Fatal - Decrypt Error
2-
tls_dtls_connection:gen_handshake/4:696: TLS server: In state certify at tls_dtls_connection.erl:696 generated SERVER ALERT: Fatal - Handshake Failure, - malformed_handshake_data
3-
ssl_record:decipher_aead/5:452: TLS server: In state connection at ssl_record.erl:452 generated SERVER ALERT: Fatal - Bad Record MAC, - decryption_failed

Iam not sure how to troubleshoot the problem as its not consistent and unfortunately GSM is mandatory in our project.
can anyone help I have been struggling for a few weeks now.

mobizt, 2024-08-24:
This is your GSM module issue.

You should provide it stable power supply.

closed 2024-08-24 by mobizt

SherifMEldeeb, 2024-08-25:
> This is your GSM module issue.
> 
> You should provide it stable power supply.

why did you assume that it's a power supply issue my power is stable and verified 
12V, 4A Adapter -> LM2596 3A Buck converter. (3.8v)  ->. Sim800l 
?!!

mobizt, 2024-08-25:
That's my experiences.

mobizt, 2024-08-25:
Use the oscilloscope for ripple voltage checking on your GSM module.