mobizt/ESP_SSLClient #20: readStringUntil() blocks execution of another thread

ikorman, 2025-03-31:
Hi

I have question regarding readStringUntil() method that in some cases blocks other tasks (in this case looptask on core 1). Here is a short setup: 

- I have dedicated task (core1) that is talking to Firebase Relatime DB
- there is set method used for sending requests to FB RTDB
- in some cases for some reason exception is thrown

```
[E (196643) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (196643) task_wdt:  - loopTask (CPU 1)
E (196643) task_wdt: Tasks currently running:
E (196643) task_wdt: CPU 0: IDLE0
E (196643) task_wdt: CPU 1: FirebaseTask
E (196643) task_wdt: Print CPU 1 backtrace

Backtrace: 0x40377103:0x3FC9B1D0 0x40379A79:0x3FC9B1F0 0x4203BE53:0x3FCB2770 0x42038DEF:0x3FCB2790 0x42038E99:0x3FCB27B0 0x420089F1:0x3FCB27D0 0x42008C22:0x3FCB2A60 0x4037F8D6:0x3FCB2AC0
  #0  0x40377103 in esp_crosscore_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/crosscore_int.c:73
  #1  0x3FC9B1D0 in _xt_exception_table at ??:?
  #2  0x40379A79 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/xtensa/xtensa_vectors.S:1240
  #3  0x3FC9B1F0 in _xt_exception_table at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:67
  #4  0x4203BE53 in millis at C:/Users/Ivan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:206 (discriminator 1)
  #5  0x42038DEF in Stream::timedRead() at C:/Users/Ivan/.platformio/packages/framework-arduinoespressif32/cores/esp32/Stream.cpp:33
  #6  0x42038E99 in Stream::readStringUntil(char) at C:/Users/Ivan/.platformio/packages/framework-arduinoespressif32/cores/esp32/Stream.cpp:244
  #7  0x420089F1 in FirebaseClass::set(String const&, String const&, FirebaseClass::Operation) at src/Firebase.cpp:360
  #8  0x42008C22 in FirebaseClass::firebaseTask(void*) at src/Firebase.cpp:255 (discriminator 4)
  #9  0x4037F8D6 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139]([url](url))
```

From what I can see it's triggered by readStringUntil(). It looks like it blocks execution of loopTask. I would appreciate if you have any idea what could be wrong or is there non-blocking way of reading incoming data.

Here are running tasks:

```
===============================================================
Task Name         Core       State     Prio   Stack(HSW)   Num
---------------------------------------------------------------
loopTask          Core 1     Running   1      5456         8
IDLE1             Core 1     Ready     0      344          6
IDLE0             Core 0     Ready     0      244          5
tiT               Core 0     Blocked   18     2868         13
WiFiHandlerTask   Core 1     Blocked   10     1360         17
SdLoggerTask      Core 1     Blocked   10     1500         10
PumpSchedulerTa   Core 1     Blocked   8      2032         23
WellPumpTask      Core 1     Blocked   8      2088         12
PulseMonitorTas   Core 1     Blocked   5      2152         22
PulseMonitorTas   Core 1     Blocked   5      2012         21
PulseMonitorTas   Core 1     Blocked   5      2148         20
MeteoUploadTask   Core 1     Blocked   1      1072         18
FirebaseTask      Core 1     Blocked   12     2680         11
ipc1              Core 1     Suspended 24     256          2
sys_evt           Core 0     Blocked   20     2732         14
arduino_events    Core 1     Blocked   19     1712         15
esp_timer         Core 0     Suspended 22     7704         3
async_tcp         Core 1     Blocked   12     12672        19
wifi              Core 0     Blocked   23     4108         16
Tmr Svc           Unpinned   Blocked   1      2412         7
ipc0              Core 0     Suspended 24     256          1
===============================================================
```

Here is my set() method

```
/// @brief Sets data at a specific path in the Firebase Realtime Database.
/// This function sends an HTTPS request to the Firebase server based on the specified operation.
/// @param path The path in the database where the operation will be performed.
/// @param jsonData The JSON data to store.
/// @param operation The type of operation to perform (PUT, POST, PATCH, DELETE).
/// @return true if the operation was successfully performed, false otherwise.
bool FirebaseClass::set(const String &path, const String &jsonData, Operation operation)
{
    if (WiFi.status() != WL_CONNECTED)

    {
        ESP_LOGE(tag, "Set failed, no WiFi connected");
        return false;
    }

    // Take the mutex to ensure exclusive access to the client
    if (xSemaphoreTake(_mutex, portMAX_DELAY) != pdTRUE)
    {
        ESP_LOGE(tag, "Failed to take mutex for set operation.");
        return false;
    }

    // Ensure the path is properly encoded
    String encodedPath = urlEncode(path);
    String fullPath = encodedPath + ".json?auth=" + _dbSecretKey;

    // Use WiFiClientSecure for HTTPS
    if (!sslClient.connected())
    {
        ESP_LOGD(tag, "Connecting to '%s'", _host.c_str());
        if (!sslClient.connect(_host.c_str(), 443))
        {
            ESP_LOGE(tag, "Connection to %s failed!", _host.c_str());
            xSemaphoreGive(_mutex);
            return false;
        }
    }

    // Determine HTTP method
    const char *httpMethod = operationToString(operation);
    if (strcmp(httpMethod, "UNKNOWN") == 0)
    {
        ESP_LOGE(tag, "Unsupported operation");
        xSemaphoreGive(_mutex);
        return false;
    }

    // Prepare HTTP request using char buffers to avoid fragmentation
    char requestHeader[512];
    int requestLength = snprintf(requestHeader, sizeof(requestHeader),
                                 "%s %s HTTP/1.1\r\n"
                                 "Host: %s\r\n"
                                 "User-Agent: ESP32\r\n"
                                 "Content-Type: application/json\r\n"
                                 "Content-Length: %d\r\n"
                                 "Connection: keep-alive\r\n\r\n",
                                 httpMethod, fullPath.c_str(), _host.c_str(), jsonData.length());

    if (requestLength <= 0 || requestLength >= (int)sizeof(requestHeader))
    {
        ESP_LOGE(tag, "Request header overflow");
        sslClient.stop();
        xSemaphoreGive(_mutex);
        return false;
    }

    uint32_t startTime = millis();
    // Send request
    sslClient.print(requestHeader);
    sslClient.print(jsonData);
    sslClient.print("\r\n");

    ESP_LOGD(tag, "HTTP request\r\n%s\r\n%s", requestHeader, jsonData.c_str());

    // Check for HTTP response status
    bool isSuccess = false;
    while (sslClient.connected())
    {
        String line = sslClient.readStringUntil('\n');
        if (line.startsWith("HTTP/1.1 200 OK") || line.startsWith("HTTP/1.1 201 Created"))
        {
            isSuccess = true;
        }
        if (line == "\r")
        {
            break;
        }
    }

    String response = "";
    while (sslClient.available())
    {
        response += sslClient.readString();
    }
    uint32_t duration = millis() - startTime;

    ESP_LOGD(tag, "HTTP response\r\n%s", response.c_str());

    if (isSuccess)
    {
        ESP_LOGI(tag, "Request succeeded > %s %s (%lu ms)", httpMethod, path.c_str(), duration);
    }
    else
    {
        ESP_LOGE(tag, "Request failed > %s %s, response: %s (%lu ms)", httpMethod, path.c_str(), response.c_str(), duration);
    }

    // Release the mutex
    xSemaphoreGive(_mutex);

    return isSuccess;
}
```

mobizt, 2025-03-31:
Ther is no such function i.e. readStringUntil in ESP_SSLClient library.

closed 2025-03-31 by mobizt

ikorman, 2025-03-31:
But what is than this?

[image]

mobizt, 2025-03-31:
That is the function from the base (Stream) class of Client class.

All network and SSL clients that are running in Arduino framework derived from the Arduino Client class (actually Stream class).

mobizt, 2025-03-31:
This is all functions that are available in this library.

https://github.com/mobizt/ESP_SSLClient/blob/main/src/client/BSSL_TCP_Client.h

Calling the base class's functions is your own risk and it does not maintain by the client library that derives from it unless the ESP32 core team.

ikorman, 2025-03-31:
OK, thx. Will have to find solution somewhere else.

mobizt, 2025-03-31:
When you use `WiFiClient` as the basic client. The issue was there.

ESP32 `WIFIClient` has the known bug from implementation in my other libraries, and no one post the issue in the ESP32 repo because of it is rarely used.

Any way you can try this bug fixed implementation of [WiFiClient](https://github.com/mobizt/ESP-Mail-Client/blob/master/src/client/WiFiClientImpl.h) but it is your own risk.

ikorman, 2025-03-31:
Thx. I was not aware of that issue, will take a look at your fix.

BR
Ivan

mobizt, 2025-03-31:
It is the core code from my library, and it works fine.
https://github.com/mobizt/ESP-Mail-Client

ikorman, 2025-03-31:
I did not say that something is wrong, just saying i have issue using it. I made dummy project to test this. First I use WiFiClient and 2nd time WiFiClientImpl

Here is code (WiFiClient):

```
#include <Arduino.h>
#include <WiFi.h>
#include <ESP_SSLClient.h>
#include <WiFiClientImpl.h>

ESP_SSLClient sslClient;
WiFiClient wifiClient;

void setup()
{
	esp_log_level_set("*", ESP_LOG_DEBUG);

	Serial.begin(115200);

	WiFi.enableAP(false);
	WiFi.enableSTA(true);
	WiFi.mode(WIFI_MODE_STA);
	WiFi.setAutoReconnect(true);
	WiFi.begin ("xxxx", "xxxx");
	
	delay (1000);

	sslClient.setInsecure(); // For testing purposes only, not secure for production
	sslClient.setBufferSizes(2048 /* rx */, 2048 /* tx */);
	sslClient.setDebugLevel(4);
	sslClient.setSessionTimeout(300);
	sslClient.setClient(&wifiClient);
	sslClient.connect ("xxxxxxx-default-rtdb.europe-west1.firebasedatabase.app", 443);
	
	Serial.println ("Setup done");	
}

void loop()
{
}
```

Output with WiFiClient is as expected:

```
W (1072) ARDUINO: Reason: 202 - AUTH_FAIL
D (1072) ARDUINO: WiFi Reconnect Running
W (1072) ARDUINO: STA already disconnected.
D (2055) ARDUINO: Clearing DNS cache
D (2070) ARDUINO: DNS found IPv4 34.107.226.223
> INFO.mConnectBasicClient: Basic client connected!
> INFO.mConnectSSL: Start connection.
> INFO.mConnectSSL: Wait for SSL handshake.
> INFO.mUpdateEngine: State RECVREC
> INFO.mUpdateEngine: State RECVREC
> INFO.mRunUntil: SSL state changed.
> INFO.mRunUntil: State RECVREC
> INFO.mRunUntil: Expected bytes count: 5
> INFO.mUpdateEngine: State RECVREC
> INFO.mUpdateEngine: State RECVREC
> INFO.mUpdateEngine: State SENDAPP
> INFO.mRunUntil: SSL state changed.
> INFO.mRunUntil: State SENDAPP
> INFO.mConnectSSL: Connection successful!
Setup done
```

When I change (and only that) WiFiClient wifiClient; to WiFiClientImpl wifiClient;, output is following:

```
W (1068) ARDUINO: Reason: 202 - AUTH_FAIL
D (1068) ARDUINO: WiFi Reconnect Running
W (1068) ARDUINO: STA already disconnected.
E (20503) ARDUINO: connect on fd 48, errno: 113, "Software caused connection abort"
> INFO.mConnectBasicClient: Basic client connected!
> INFO.mConnectSSL: Start connection.
> INFO.mConnectSSL: Wait for SSL handshake.
> INFO.mUpdateEngine: State RECVREC
> ERROR.mUpdateEngine: Error writing to basic client.
> WARN.mRunUntil: Terminating with write error.
> ERROR.mConnectSSL: Failed to initlalize the SSL layer.
> ERROR.mConnectSSL: Unknown error code.
Setup done
```

Here is content of platformio.ini:

```
[env:stable]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = esp32-s3-devkitc-1-n16r8v
framework = arduino
monitor_speed = 115200
monitor_filters = 
	esp32_exception_decoder
	direct
	log2file
build_flags = 
	-D CORE_DEBUG_LEVEL=5
	-D USE_ESP_IDF_LOG=1    
lib_deps =     
	mobizt/ESP_SSLClient@^2.1.16   
``` 

Output from compile:

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1-n16r8v.html
PLATFORM: Espressif 32 (53.3.13) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.1.3
 - framework-arduinoespressif32-libs @ 5.3.0+sha.489d7a2b3a
 - tool-esptoolpy @ 4.8.6
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-riscv32-esp @ 13.2.0+20240530
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP_SSLClient @ 2.1.16
|-- WiFi @ 3.1.3
Building in release mode
```

mobizt, 2025-03-31:
I just to clarify that something wrong in replacement.

I can successfully connect to any hosts.

For Firebase host, sometimes the ESP32 core dns resolver was failed.

Actually, you should get error from ESP32 debug.

`[  7433][E][NetworkManager.cpp:136] hostByName(): DNS Failed for 'xxxxx-default-rtdb.xxxx-xxxxx.firebasedatabase.app' with error '-54'`

Before 
`> ERROR.mUpdateEngine: Error writing to basic client.`

The WiFiClientImpl is actually ESP32 v2.0.x WiFiClient class with the issue fix and it uses the  ESP32 WiFi (v2.0.x) and NetworkManager (v3.x.x) classes's hostByName() function to resolve the domain name.

Anyway you are using out dated platformIO ESP32 sdk which no longer maintained
`platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip`

Use the latest
`platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.20/platform-espressif32.zip`

The issue is actually up on the network condition not because WiFiClient you are using.

WiFiClientImpl is actually soved the known issue, see the file comment.

mobizt, 2025-03-31:
And it is up on you as the issue only existed in the buggy `ESP32 WiFiClient`.

ikorman, 2025-03-31:
I'm brain dead... Installed 54.03.20 (after reinstalling PIO from scratch + deleting .platformio folder), same as before... Need some time to relax and have fresh brain as I have no idea what is wrong in my case

I used file from: github.com/mobizt/ESP-Mail-Client/blob/master/src/client/WiFiClientImpl.h

```
============ Before Setup End ============
W (1110) ARDUINO: Reason: 202 - AUTH_FAIL
D (1110) ARDUINO: WiFi Reconnect Running
W (1110) ARDUINO: STA already disconnected.
E (21495) ARDUINO: connect on fd 48, errno: 113, "Software caused connection abort"
> INFO.mConnectBasicClient: Basic client connected!
> INFO.mConnectSSL: Start connection.
> INFO.mConnectSSL: Wait for SSL handshake.
> INFO.mUpdateEngine: State RECVREC
> ERROR.mUpdateEngine: Error writing to basic client.
> WARN.mRunUntil: Terminating with write error.
> ERROR.mConnectSSL: Failed to initlalize the SSL layer.
> ERROR.mConnectSSL: Unknown error code.
Setup done
=========== After Setup Start ============
```

```
[env:stable]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20/platform-espressif32.zip
board = esp32-s3-devkitc-1-n16r8v
framework = arduino
monitor_speed = 115200
monitor_filters = 
	esp32_exception_decoder
	direct
	log2file
build_flags = 
	-D CORE_DEBUG_LEVEL=5
	-D USE_ESP_IDF_LOG=1    
lib_deps =     
	mobizt/ESP_SSLClient@^2.1.16    
```

```
#include <Arduino.h>
#include <WiFi.h>
#include <ESP_SSLClient.h>
#include <WiFiClientImpl.h>

ESP_SSLClient sslClient;
WiFiClientImpl wifiClient;

void setup()
{
	esp_log_level_set("*", ESP_LOG_DEBUG);

	Serial.begin(115200);

	WiFi.enableAP(false);
	WiFi.enableSTA(true);
	WiFi.mode(WIFI_MODE_STA);
	WiFi.setAutoReconnect(true);
	WiFi.begin ("xxxxxx", "xxxxxxxxx");
	
	delay (2000);

	sslClient.setInsecure(); // For testing purposes only, not secure for production
	sslClient.setBufferSizes(2048 /* rx */, 2048 /* tx */);
	sslClient.setDebugLevel(4);
	sslClient.setSessionTimeout(300);
	sslClient.setClient(&wifiClient);
	sslClient.connect ("xxxxxxxxx-default-rtdb.europe-west1.firebasedatabase.app", 443);
	
	Serial.println ("Setup done");	
}

void loop()
{
}

```

```
PLATFORM: Espressif 32 (54.3.20) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.2.0
 - framework-arduinoespressif32-libs @ 5.4.0+sha.2f7dcd862a
 - tool-esptoolpy @ 4.8.8
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-riscv32-esp @ 14.2.0+20241119
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP_SSLClient @ 2.1.16
|-- WiFi @ 3.2.0
Building in release mode
```

mobizt, 2025-03-31:
Ok I know the problem of `WiFiClientImpl` now.

Please use the following code of `WiFiClientImpl.h` instead.

https://gist.github.com/mobizt/6a8e7259d1ccd15a2b2ce62b604dd7ff

The problem is some undefined macro guards the dns resolver to work.

ikorman, 2025-03-31:
Great, finally getting somewhere :) Now it seems to be working with WiFiClientImpl, here is output:

```
============ Before Setup End ============
W (1063) ARDUINO: Reason: 202 - AUTH_FAIL
D (1063) ARDUINO: WiFi Reconnect Running
W (1063) ARDUINO: STA already disconnected.
D (3043) ARDUINO: Clearing DNS cache
D (4041) ARDUINO: DNS found IPv4 34.107.226.223
> INFO.mConnectBasicClient: Basic client connected!
> INFO.mConnectSSL: Start connection.
> INFO.mConnectSSL: Wait for SSL handshake.
> INFO.mUpdateEngine: State RECVREC
> INFO.mUpdateEngine: State RECVREC
> INFO.mRunUntil: SSL state changed.
> INFO.mRunUntil: State RECVREC
> INFO.mRunUntil: Expected bytes count: 5
> INFO.mRunUntil: Expected bytes count: 2368
> INFO.mUpdateEngine: State RECVREC
> INFO.mUpdateEngine: State RECVREC
> INFO.mRunUntil: Expected bytes count: 5
> INFO.mUpdateEngine: State SENDAPP
> INFO.mRunUntil: SSL state changed.
> INFO.mRunUntil: State SENDAPP
> INFO.mConnectSSL: Connection successful!
Setup done
=========== After Setup Start ============
```

I will tomorrow take this change and implement it into main app to see how the app will behave, will let you know of outcome. 

Thank you for your support. 👍

ikorman, 2025-04-01:
Well, problem is that I can not use this with HttpClient lib as it's expecting client as NetworkClient. Let's close this topic, as this is going outside initial problem.

Thx for the support. Maybe in the future I will switch to your Firebase lib, but at the moment I will try rework my existing implementation based on HttpClient