mobizt/ESP_SSLClient #23: Works flawlessly with AWS labda endpoint, but Exception (4) crash with local HTTPS server

yas-punathil, 2025-05-13:
Hi,

I used this library to implement HTTPS client to send data packets to AWS lambda endpoint. It is working fine. But for testing purpose, when I changed the endpoint to local HTTPS server implemented in python, the ESP8266 code crashes after few (randomly) successful data transmission. 

I am always getting exception(4) like the trace given below, and when I decoded the trace it always point to String realloc(). And I don't have heap fragmentation. I verified the maximum free heap block to be 15KB just before it crashes.

Exception trace
2025-05-13 11:54:48,060 - INFO - b'Soft WDT reset\r\n'
2025-05-13 11:54:48,060 - INFO - b'\r\n'
2025-05-13 11:54:48,060 - INFO - b'Exception (4):\r\n'
2025-05-13 11:54:48,060 - INFO - b'epc1=0x4021601f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000\r\n'
2025-05-13 11:54:48,062 - INFO - b'\r\n'
2025-05-13 11:54:48,062 - INFO - b'>>>stack>>>\r\n'
2025-05-13 11:54:48,062 - INFO - b'\r\n'
2025-05-13 11:54:48,062 - INFO - b'ctx: cont\r\n'
2025-05-13 11:54:48,062 - INFO - b'sp: 3ffffd70 end: 3fffffd0 offset: 0160\r\n'

The code snippet below indicates how I use the library in my ESP8266 code:

ESP_SSLClient ssl_client;
static WiFiClient wifi_client;  // Persistent WiFiClient for the HTTP session

/**
 * @brief Initialize the HTTP client with the specified configuration.
 */
bool http_client_init(http_client_t* client, const http_client_config_t* config) {
    if (!client || !config) {
        return false;
    }
    memcpy(&client->config, config, sizeof(http_client_config_t));
    client->user_context = NULL;

    // Set SSL client to operate in insecure mode 
    ssl_client.setInsecure();

    // Set buffer sizes for SSL client
    // Parameters: receive buffer size, send buffer size
    ssl_client.setBufferSizes(SSL_CLIENT_RX_BUF_SIZE, SSL_CLIENT_TX_BUF_SIZE);
    // ssl_client.setCACert(root_ca);  // Set the certificate

    // Set debug level for SSL client
    // Parameter: debug level (0 for no debug info, 1 for basic debug info, higher levels for more verbose output)
    // ssl_client.setDebugLevel(3);

    // ssl_client.setTimeout(3);       // 

    // Set underlying TCP client for SSL client
    // Parameter: pointer to basic TCP client object
    ssl_client.setClient(&wifi_client);
    
    return true;
}

/**
 * @brief start http client stream.
 */
int http_client_start_stream(http_client_t* client, const char* initial_payload_chunk, size_t data_size) {
    if (!client || !initial_payload_chunk) {
        return false;
    }

    size_t initial_chunk_length = strlen(initial_payload_chunk);
    size_t content_length = initial_chunk_length + data_size + strlen("\"}");  // closing content

    const char* host = client->config.server_endpoint;
    uint16_t port = client->config.port;
    const char* path = client->config.api_path;

    BSQ_DEBUG_VAL("host: ", host);
    BSQ_DEBUG_VAL("port: ", port);

    if (!ssl_client.connect(host, port)) {
        BSQ_DEBUG_PRINT("[http] Connection failed");
        return false;
    }

    // --- Construct the HTTP headers manually ---
    char header[512];  // Adjust if you need more
    snprintf(header, sizeof(header),
             "POST %s HTTP/1.1\r\n"
             "Host: %s\r\n"
             "Content-Type: application/json\r\n"
             "Content-Length: %u\r\n"
             "Connection: keep-alive\r\n"
             "\r\n",
             path, host, (unsigned int)content_length);

    // --- Send headers ---
    size_t len = strlen(header);
    if(ssl_client.write((const uint8_t*)header, len) != len){
        return false;
    }

    // --- Send the initial chunk of the body ---
    if(ssl_client.write((const uint8_t*)initial_payload_chunk, initial_chunk_length) != initial_chunk_length){
        return false;
    }

    // Do not close the connection — caller will continue writing
    return true;
}

/**
 * @brief Stream a single data block to the server.
 */
bool http_client_stream_block(http_client_t* client, const char* data, size_t length) {
    if (!client || !data || length == 0) {
        return false;
    }

    // Use the same global or passed-in WiFiClient as in start_stream
    if (!ssl_client.connected()) {
        BSQ_DEBUG_PRINT("[http] Stream not connected");
        return false;
    }

    size_t written = ssl_client.write((const uint8_t*)data, length);

    if (written != length) {
        BSQ_DEBUG_PRINT("[http] Partial write: wrote");
        return false;
    }

    return true;
}

/**
 * @brief End the HTTP chunked transfer session.
 */
bool http_client_end_stream(http_client_t* client) {
    if (!client) {
        return false;
    }

    if (!ssl_client.connected()) {
        BSQ_DEBUG_PRINT("[http] Client already disconnected");
        return false;
    }

    // read the response line
    String response_line = ssl_client.readStringUntil('\n');
    // BSQ_DEBUG_PRINT("[http] Response: " + response_line);

    // Close the connection
    ssl_client.stop();
    return true;
}

The below is my python server code:

#storing data into folder structure based on unique_id
import os
import json
import ssl
import platform
from http.server import BaseHTTPRequestHandler, HTTPServer

class LambdaLikeHandler(BaseHTTPRequestHandler):
    def do_POST(self):
        try:
            content_length = int(self.headers.get('Content-Length', 0))
            raw_data = self.rfile.read(content_length)
            print(f"[SERVER] Raw data: {raw_data}")
            data = raw_data.decode('utf-8', errors='replace')
            print("[SERVER] Raw data received.")

            try:
                decoded_data = json.loads(data)
            except json.JSONDecodeError:
                print("Failed to decode JSON.")
                self.send_response(400)
                self.end_headers()
                return

            unique_id = decoded_data.get('unique_id')
            if not unique_id:
                print("No 'unique_id' found in the data.")
                self.send_response(400)
                self.end_headers()
                return

            data_content = decoded_data.get('data')
            if not data_content:
                print("No 'data' field found in the JSON.")
                self.send_response(400)
                self.end_headers()
                return

            # Determine OS-safe path
            base_path = '.' if platform.system() == 'Windows' else '/home/bsq/Public/AC_data'

            # Choose subfolder based on path
            if self.path == '/stream/compressed':
                folder_path = os.path.join(base_path, unique_id, 'compressed')
            elif self.path == '/stream/uncompressed':
                folder_path = os.path.join(base_path, unique_id, 'uncompressed')
            else:
                print(f"Unknown path: {self.path}")
                self.send_response(400)
                self.end_headers()
                return

            # Create directory if it doesn't exist
            os.makedirs(folder_path, exist_ok=True)
            print(f"[SERVER] Using folder: {folder_path}")

            # Find the next file number
            packet_number = 1
            while os.path.exists(os.path.join(folder_path, f"packet_{packet_number}.txt")):
                packet_number += 1

            file_path = os.path.join(folder_path, f"packet_{packet_number}.txt")
            with open(file_path, 'w') as f:
                f.write(data_content)

            print(f"[SERVER] Data saved to {file_path}")

            # Respond back
            self.send_response(200)
            self.send_header('Content-Type', 'application/json')
            self.end_headers()
            self.wfile.write(b'{"message": "Data received and saved successfully."}')

        except Exception as e:
            print(f"[SERVER] Exception occurred: {e}")
            self.send_response(500)
            self.end_headers()

    def log_message(self, format, *args):
        return  # Suppress default logging

def run_server(port=443, cert_file='Iam2.pem', key_file='Iam1.pem'):
    server_address = ('0.0.0.0', port)
    httpd = HTTPServer(server_address, LambdaLikeHandler)
    
    # Wrap with SSL using context (modern way)
    # context = ssl.SSLContext(ssl.PROTOCOL_TLS_SERVER)
    context = ssl.SSLContext(ssl.PROTOCOL_TLSv1_2)
    context.options |= ssl.OP_NO_TLSv1_3  # Disable TLS 1.3
    context.set_ciphers("ECDHE-RSA-AES128-GCM-SHA256")  # Compatible with ESP8266
    context.load_cert_chain(certfile=cert_file, keyfile=key_file)
    httpd.socket = context.wrap_socket(httpd.socket, server_side=True)

    print(f"[SERVER] Listening on HTTPS port {port} (simulating AWS Lambda)")
    # httpd.socket.settimeout(5)
    httpd.serve_forever()

if __name__ == '__main__':
    run_server()

closed 2025-05-13 by mobizt

yas-punathil, 2025-05-13:
Hi, I replicated the issue with the HTTPS example code in the repository and the given python server code.

mobizt, 2025-05-13:
That is your own application, and I cannot test it for you.

yas-punathil, 2025-05-13:
I tested with WiFiClientSecure.h, and it is not crashing like I experienced with ESP_SSLClient.h, with the same server code.