mobizt/ESP_SSLClient #27: Memory leak when loading CACert?

Bascy, 2025-06-21:
In our project we only need SSLClients for a very limited amount of time, so we create and destroy them as needed.
When using the `sslClient.setInsecure()` then there is no problem, but when loading a certificate then there is memory leaking when we destroy the sslclient. I'm assuming that the loaded certificate data is not released with the ssl_client...

```c++
#include <Arduino.h>
#include <ESP_SSLClient.h>

// ISRG Root X1
// Expired on Fri Jun 07 11:06:39 UTC 2030
const char rootCA[] PROGMEM = "-----BEGIN CERTIFICATE-----\n"
...
                              "-----END CERTIFICATE-----\n";

void makeSSLClient()
{

  ESP_SSLClient ssl_client;
  Serial.printf("Client created at %p: %d\n", &ssl_client, ESP.getFreeHeap());
  ssl_client.setCACert(rootCA);
  Serial.printf("SSL Client with CA Cert, freemem: %d\n", ESP.getFreeHeap());
}

void setup()
{
  Serial.begin(576000);
}

void loop()
{
  delay(10000);
  Serial.println("------------------------");
  Serial.printf("FreeMem before: %d\n", ESP.getFreeHeap());
  makeSSLClient();
  Serial.printf("FreeMem after: %d\n", ESP.getFreeHeap());
}
```

output:
```
FreeMem before: 286144
Client created at 0x3ffb20e0: 286144
SSL Client with CA Cert, freemem: 283956
FreeMem after: 283956
------------------------
FreeMem before: 283956
Client created at 0x3ffb20e0: 283956
SSL Client with CA Cert, freemem: 281764
FreeMem after: 281764
------------------------
FreeMem before: 281764
Client created at 0x3ffb20e0: 281764
SSL Client with CA Cert, freemem: 279564
FreeMem after: 279564
------------------------
FreeMem before: 279564
Client created at 0x3ffb20e0: 279564
SSL Client with CA Cert, freemem: 277384
FreeMem after: 277384
------------------------
FreeMem before: 277384
Client created at 0x3ffb20e0: 277384
SSL Client with CA Cert, freemem: 275192
FreeMem after: 275192
------------------------
FreeMem before: 275192
Client created at 0x3ffb20e0: 275192
SSL Client with CA Cert, freemem: 272996
FreeMem after: 272996
```

Bascy, 2025-06-21:
The problem can be solved by adding `mClearAuthenticationSettings();` to the `BSSL_SSL_Client()` destructor

```c++
BSSL_SSL_Client::~BSSL_SSL_Client()
{
    if (_basic_client)
    {
        if (_basic_client->connected())
            _basic_client->stop();
        _basic_client = nullptr;
    }
    freeImpl(&_cipher_list);
    mFreeSSL();
    mClearAuthenticationSettings();
#if defined(USE_EMBED_SSL_ENGINE)
    stack_thunk_del_ref();
#endif
}

mobizt, 2025-06-22:
The issue was fixed in [v2.2.3](https://github.com/mobizt/ESP_SSLClient/releases/tag/v2.2.3). Please update.

closed 2025-06-22 by mobizt

Bascy, 2025-06-22:
Thank you for the rapid response!